Heltec-Aaron-Lee/WiFi_Kit_series #168: LoRaWAN_OLED Wireless Stick v3 compiling error / not supported

fhb, 2023-02-22:
The example script LoRaWAN_OLED does not support the Wireless Stick(V3) = "Wireless_Stick_V3", probably as the files have to be adapted to the smaller display (64*32). LoRaWAN_APP.h and LoRaWAN_APP.cpp only define the display if it's a WIFI_Lora_32 device.

Error message when compiling LoRaWAN_OLED (framework 0.0.7, library 1.1.1):

```
/private/var/folders/br/k52zl5l16bgg65qt6cqzmm9r0000gn/T/.arduinoIDE-unsaved2023122-3535-1xx1337.gijvh/LoRaWan_OLED/LoRaWan_OLED.ino: In function 'void setup()':
/private/var/folders/br/k52zl5l16bgg65qt6cqzmm9r0000gn/T/.arduinoIDE-unsaved2023122-3535-1xx1337.gijvh/LoRaWan_OLED/LoRaWan_OLED.ino:83:13: error: 'class LoRaWanClass' has no member named 'displayMcuInit'
     LoRaWAN.displayMcuInit();
             ^~~~~~~~~~~~~~
/private/var/folders/br/k52zl5l16bgg65qt6cqzmm9r0000gn/T/.arduinoIDE-unsaved2023122-3535-1xx1337.gijvh/LoRaWan_OLED/LoRaWan_OLED.ino: In function 'void loop()':
/private/var/folders/br/k52zl5l16bgg65qt6cqzmm9r0000gn/T/.arduinoIDE-unsaved2023122-3535-1xx1337.gijvh/LoRaWan_OLED/LoRaWan_OLED.ino:103:15: error: 'class LoRaWanClass' has no member named 'displayJoining'
       LoRaWAN.displayJoining();
               ^~~~~~~~~~~~~~
/private/var/folders/br/k52zl5l16bgg65qt6cqzmm9r0000gn/T/.arduinoIDE-unsaved2023122-3535-1xx1337.gijvh/LoRaWan_OLED/LoRaWan_OLED.ino:109:15: error: 'class LoRaWanClass' has no member named 'displaySending'
       LoRaWAN.displaySending();
               ^~~~~~~~~~~~~~
/private/var/folders/br/k52zl5l16bgg65qt6cqzmm9r0000gn/T/.arduinoIDE-unsaved2023122-3535-1xx1337.gijvh/LoRaWan_OLED/LoRaWan_OLED.ino:125:15: error: 'class LoRaWanClass' has no member named 'displayAck'
       LoRaWAN.displayAck();
               ^~~~~~~~~~

exit status 1

Compilation error: 'class LoRaWanClass' has no member named 'displayMcuInit'
```

LoRaWAN_APP.h :
```
#if defined(WIFI_LoRa_32_V3)||defined(WIFI_LoRa_32_V2))
  void displayJoining();
  void displayJoined();
  void displaySending();
  void displayAck();
  void displayMcuInit();
#endif
};
```
LoRaWAN_APP.ccp
```
#if defined(WIFI_LoRa_32_V3)||defined(WIFI_LoRa_32_V2)||defined(WIFI_LoRa_32)
#include <Wire.h>  
#include "HT_SSD1306Wire.h"

  SSD1306Wire  display(0x3c, 500000, SDA_OLED, SCL_OLED, GEOMETRY_128_64, RST_OLED);; // addr , freq , i2c group , resolution , rst

  uint8_t ifDisplayAck=0;
  uint8_t isDispayOn=0;
#endif
```

if I change the resolution and add defined(Wireless_Stick_V3) I get the following error:
```
/Users/Felix/Library/Arduino15/packages/Heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/k52zl5l16bgg65qt6cqzmm9r0000gn/T/arduino-sketch-8B13E11C0387F3067B9BA1273BDB61C5/sketch/LoRaWan_OLED_custom.ino.cpp.o:(.literal._Z5setupv+0x20): undefined reference to `LoRaWanClass::displayMcuInit()'
/Users/Felix/Library/Arduino15/packages/Heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/k52zl5l16bgg65qt6cqzmm9r0000gn/T/arduino-sketch-8B13E11C0387F3067B9BA1273BDB61C5/sketch/LoRaWan_OLED_custom.ino.cpp.o:(.literal._Z4loopv+0x18): undefined reference to `LoRaWanClass::displayAck()'
/Users/Felix/Library/Arduino15/packages/Heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/k52zl5l16bgg65qt6cqzmm9r0000gn/T/arduino-sketch-8B13E11C0387F3067B9BA1273BDB61C5/sketch/LoRaWan_OLED_custom.ino.cpp.o:(.literal._Z4loopv+0x1c): undefined reference to `LoRaWanClass::displayJoining()'
/Users/Felix/Library/Arduino15/packages/Heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/k52zl5l16bgg65qt6cqzmm9r0000gn/T/arduino-sketch-8B13E11C0387F3067B9BA1273BDB61C5/sketch/LoRaWan_OLED_custom.ino.cpp.o:(.literal._Z4loopv+0x20): undefined reference to `LoRaWanClass::displaySending()'
/Users/Felix/Library/Arduino15/packages/Heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/k52zl5l16bgg65qt6cqzmm9r0000gn/T/arduino-sketch-8B13E11C0387F3067B9BA1273BDB61C5/sketch/LoRaWan_OLED_custom.ino.cpp.o: in function `setup()':
/Users/Felix/Documents/Arduino/LoRaWan_OLED_custom/LoRaWan_OLED_custom.ino:81: undefined reference to `LoRaWanClass::displayMcuInit()'
/Users/Felix/Library/Arduino15/packages/Heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/k52zl5l16bgg65qt6cqzmm9r0000gn/T/arduino-sketch-8B13E11C0387F3067B9BA1273BDB61C5/sketch/LoRaWan_OLED_custom.ino.cpp.o: in function `loop()':
/Users/Felix/Documents/Arduino/LoRaWan_OLED_custom/LoRaWan_OLED_custom.ino:125: undefined reference to `LoRaWanClass::displayAck()'
/Users/Felix/Library/Arduino15/packages/Heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/Felix/Documents/Arduino/LoRaWan_OLED_custom/LoRaWan_OLED_custom.ino:103: undefined reference to `LoRaWanClass::displayJoining()'
/Users/Felix/Library/Arduino15/packages/Heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/Felix/Documents/Arduino/LoRaWan_OLED_custom/LoRaWan_OLED_custom.ino:105: undefined reference to `LoRaWanClass::displaySending()'
/Users/Felix/Library/Arduino15/packages/Heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/br/k52zl5l16bgg65qt6cqzmm9r0000gn/T/arduino-sketch-8B13E11C0387F3067B9BA1273BDB61C5/libraries/LoraWan102/LoRaWan_APP.cpp.o:(.literal._ZN12LoRaWanClass5sleepE12eDeviceClass+0x0): undefined reference to `Radio'
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

fred777, 2023-02-28:
I can confirm this. And I already regret buying the Heltec board instead of TTGO :-/

fhb, 2023-02-28:
I posted a pull request after I found more "define"-sections and made the script runnable: https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/pull/170. You can just download and replace the changed files.

But I agree and consider trying a ttgo board and to write off the 50$ the heltec boards cost. It was quite frustrating and time intensive to try to fix the library. As I found out progressing from this, most of the scripts provided by heltec don't run properly without further adjustments and the documentation is lacking. Even the pins were not defined properly. It's lazy and even rude. These things should've been tested by professionals before selling this product!

fred777, 2023-03-02:
A look at the meshtastic code base might be worthwhile - they have full support for the Heltec LORA32 v3 board: https://github.com/meshtastic/firmware/blob/master/variants/heltec_v3/platformio.ini

usscsmith, 2024-03-08:
> I posted a pull request after I found more "define"-sections and made the script runnable: #170. You can just download and replace the changed files.
> 
> But I agree and consider trying a ttgo board writing off the 50$ for the heltec boards. It was quite frustrating and time intensive to try to fix the library. As I found out progressing from this, most of the scripts provided by heltec don't run properly without further adjustments and the documentation is lacking. Even the pins were not defined properly. It's lazy and even rude. These things should've been tested by professionals before selling this product!

I grabbed two Heltec Wireless Stick V3 Boards for my first steps into LoRaWAN...  So far this has been insanely frustrating task to learn with these boards!  Luckily a few days ago on another forum someone finally helped me a bit and pointed me over to RadioLib. I have them both up working with Sensors in my Workshop.. Still testing but I can start to see some hope now. LMAO

Getting the OLED to work just seems like a lost cause to me at the moment.

faizannazir, 2024-03-09:
> Platform.io seems to have much better support for v3 boards! Didn't have any issues since migration from Arduino IDE. You just need to figure out the proper pins.

how to configure wireless kit v3 in platformio?