Heltec-Aaron-Lee/WiFi_Kit_series #171: Updates for Heltec WiFi kit 32(V3)

clbrandin, 2023-02-24:
After playing with more ino files with the Heltec Wifi Kit 32 (V3) board it appears that you need to update the Heltec.h and Heltec.cpp files. I posted issue #164 concerning this previously and have made a small update since then. I've attached updated files:

[Heltec.zip](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/files/10826918/Heltec.zip)

clbrandin, 2023-02-25:
Here is a sketch file for a program that tests WiFi, Bluetooth, and Display functions. Instructions are at the top of the sketch file. This will help determine whether you are setup with the latest Arduino include files.

[Wifi kit 32 V3 Test.zip](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/files/10832097/Wifi.kit.32.V3.Test.zip)

platypii, 2023-02-28:
I have Pull Requests to address this issue:

 - Fixed by #167 in this repo
 - And https://github.com/espressif/arduino-esp32/pull/7877
 - And https://github.com/HelTecAutomation/Heltec_ESP32/pull/100

Hopefully they can be reviewed soon