Heltec-Aaron-Lee/WiFi_Kit_series #173: Spiffs Error

Baptou88, 2023-03-15:
Hi all!

I am facing a problem concerning data logging with a Heltec LoRa v3 that uses an Esp32S3.

I try to save data to a csv file every 30 seconds, but after a few days I get an error :

`[vfs_api.cpp:301] VFSFileImpl(): fopen(/spiffs/data.csv) failed`

If I restart my esp32 my code works fine for a few minutes and then I get the same error

Here's my code:

```cpp
bool saveDataCsV(void){
    File myFile;
	if (!SPIFFS.exists("/data.csv"))
        {
		Serial.println("Creating data.csv");
		File myFile = SPIFFS.open("/data.csv",FILE_WRITE);
                myFile.print("Date,Tachy,Niveau,CibleVanne,OuvertureVanne,Tension,Intensite,Power");                    
	    myFile.close();
	    	
        }
        myFile = SPIFFS.open("/data.csv",FILE_APPEND);
	
         myFile.print("\n"+String(timeClient.getEpochTime())+","+String(dataTurbine.tacky)+","+String(dataEtang.ratioNiveauEtang)+","+ String(dataTurbine.targetPositionVanne)+","+String(dataTurbine.positionVanne)+"," + String(dataTurbine.U)+"," + String(dataTurbine.I)+"," + String(dataTurbine.getPower()));

    myFile.close();
  
    return true;
}
```
Some hypotheses of my problem without certainty:

Is my file very big? (Currently it weighs 546 KB and 10500 lines)

is it a problem of score of memory? (My ESP memory is 66% used

SPIFFS.usedBytes() = 948027
SPIFFS.totalBytes() = 1438481

Sincerely Baptiste

PaulSchulz, 2023-03-15:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spiffs.html

I have no experience with SPIFFS itself, but from the above documentation..

   -

   SPIFFS is able to reliably utilize only around 75% of assigned partition
   space.

This may be your problem.

Baptou88, 2023-03-16:
Thanks for your reply