Heltec-Aaron-Lee/WiFi_Kit_series #187: Readme.md Link to 404 Page

PKeyes505, 2023-07-06:
The page linked in the README.md for installing from Library Manger returns a 404 error.
  
README.md
Installation Instructions
    Using Arduino IDE Boards Manager (preferred)
       Install ESP32 form Boards Manager
       

https://heltec-automation-docs.readthedocs.io/en/latest/esp32/quick_start.html#via-arduino-board-manager

Thank you everything!

ShotokuTech, 2023-08-30:
These URLs change all the time for some reason. Try:
https://docs.heltec.org/en/node/esp32/wifi_lora_32/quick_start.html#via-arduino-board-manager

mdroberts1243, 2023-09-18:
Now the updated link is broken (404) as well...  I referenced the V2 page to get going... https://docs.heltec.org/en/node/esp32/esp32_general_docs/quick_start.html#via-arduino-board-manager

Mr-HaleYa, 2024-01-10:
@PKeyes505 
Links updated please close