Heltec-Aaron-Lee/WiFi_Kit_series #199: Wrong path for esptool

FStefanni, 2023-08-24:
Hi,

for current version (0.0.8), in file `esp32/platform.txt` the line:

```
tools.esptool_py.path={runtime.platform.path}/tools/esptool
```

Should be actually:
```
tools.esptool_py.path={runtime.platform.path}/tools
```

Regards

shimniok, 2023-09-01:
I ran into the same problem. Once I fixed it, I now get the error:

```
Traceback (most recent call last):
  File "/home/mes/.arduino15/packages/Heltec-esp32/hardware/esp32/0.0.8/tools/esptool.py", line 27, in <module>
    import serial
ModuleNotFoundError: No module named 'serial'
Multiple libraries were found for "Wire.h"
 Used: /home/mes/.arduino15/packages/Heltec-esp32/hardware/esp32/0.0.8/libraries/Wire
 Not used: /home/mes/Arduino/libraries/Wire
exit status 1
Error compiling for board WiFi Kit 32(V3).
```

So the install doesn't include the serial python library or has an incorrect include path, I suspect.

FStefanni, 2023-09-04:
Hi,

I do not know why your change is partially working.
But my change works fine for me, so maybe you can try to see if it works also for you.

Regards

Mr-HaleYa, 2024-01-10:
There seems to be a bit of back and forth on this?
Why does it keep getting changed?

@fhb 
https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/commit/320f30358d8e3207717e7fa0c0762a0503652ed8
https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/commit/f789618ec9ff1534a981038745dcb403971e1638
@wangxiangwangeuse 
https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/commit/c604804d91f2a1f0b1835589c13dfdc963bd6d51