Heltec-Aaron-Lee/WiFi_Kit_series #203: WiFi_LoRa_32_V3  - Compilation error: Multiple libraries were found for "WiFi.h"

joaopedros2, 2023-08-31:
I am using the following libraries in my sketch:

```
#include "LoRaWan_APP.h"
#include "Arduino.h"
#include <WiFi.h>
#include <PubSubClient.h>
#include <Wire.h>
#include "HT_SSD1306Wire.h"
```

and the error is as follows:
```
c:/users/joaopedros2/appdata/local/arduino15/packages/heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: libraries\LoraWan102\LoRaWan_APP.cpp.o:C:\Users\joaopedros2\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\libraries\LoraWan102\src/LoRaWan_APP.cpp:25: multiple definition of `display'; sketch\LoRaReceiver_MQTT.ino.cpp.o:C:\Users\joaopedros2\Downloads\DOMOTICA\ESP32\arduino\HELTEC_WiFi_LoRa_32_V3\sketchs\LoRaReceiver_MQTT/LoRaReceiver_MQTT.ino:8: first defined here
collect2.exe: error: ld returned 1 exit status
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\joaopedros2\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Error compiling for board WiFi LoRa 32(V3) / Wireless shell(V3) / Wireless stick lite (V3).
```

I reinstalled everything again to make sure there was nothing wrong with my Arduino on Windows, and after reinstalling everything, the only library I added was PubSubClient.h
Any idea what could be causing this issue?

shimniok, 2023-09-01:
I get the same error when trying to run the DrawingDemo example. (among other issues)

```
python3: can't open file '/home/mes/.arduino15/packages/Heltec-esp32/hardware/esp32/0.0.8/tools/esptool/esptool.py': [Errno 2] No such file or directory
Multiple libraries were found for "Wire.h"
 Used: /home/mes/.arduino15/packages/Heltec-esp32/hardware/esp32/0.0.8/libraries/Wire
 Not used: /home/mes/Arduino/libraries/Wire
exit status 2
/home/mes/.local/opt/arduino-1.8.15/arduino-builder returned 2
Error compiling for board WiFi Kit 32(V3).
```

joaopedros2, 2023-09-06:
Does anyone know which bookstores work with OLED and PubSubClient.h?

AdamZiegler, 2023-09-27:
> python3: can't open file '/home/mes/.arduino15/packages/Heltec-esp32/hardware/esp32/0.0.8/tools/esptool/esptool.py': [Errno 2] No such file or directory

With this release, there needs to be a copy of esptool.py under the folder structure listed. I created the directory and copied esptool.py into it and had this error go away.

joaopedros2, 2023-10-21:
This method works in 0.0.7?