Heltec-Aaron-Lee/WiFi_Kit_series #211: WiFi_LoRa_32_V2, Arduino IDE 2.2.1, Compile Error: LoRaWan_APP.h: No such file or directory

madmacks59, 2023-09-22:
So...I have a couple of older LoRa board (V2) and I am not able to get them to running. I have the Arduino IDE (2.2.1) set up, with the libraries installed for Heltec, and when I select the example sketch (OLED_LORA_RECEIVER) and try to compile it throws an  error...

Compilation error: LoRaWan_APP.h: No such file or directory

It seems like everything installed properly, but when I do a WinDoze file scan I don't fine LoRaWan_APP.h anywhere. Shouldn't the needed libraries get installed when the board drivers are installed?

nroselnik, 2023-09-25:
I actually have the same problem,  I added https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/releases/download/0.0.7/package_heltec_esp32_index.json
to the additional board. Heltec-examples shows up for a while. Afterwards, it disappeared and I received the same error.

nroselnik, 2023-09-27:
Right, just to update. Apparently, they released version 0.08, which was auto-updated. Short-term fix: if the heltec examples did not show up in your example sketches, remove 0.08 and reinstall the 0.07. It should show up again.

FrankRenp, 2023-12-04:
It works with the Heltec ESP32 Dev-board library 1.1.1. (heltec.h included in the example).
With 1.1.2 it has been changed and the LoRaWan_APP.h is included in the example (with is "not found") up to the current 1.1.5 version.

BHLabsLtd, 2024-01-23:
Had the same issue, had to remove ESP32 in the board manager and only have WiFi Kit board added.

seems to switch between the 2 and confuses it self.

ShotokuTech, 2024-11-04:
I resolve a different problem. But you might find this useful:
https://youtu.be/c1lkIko4Wsk?si=3eKGeQ3jlG1Wu_-9