Heltec-Aaron-Lee/WiFi_Kit_series #213: AArch64 b

mpovel, 2023-09-23:


mpovel, 2023-09-23:
I am trying to run arduino-cli or rasberry pi 4 64 Bit and get errors as the host os is not supported.

dswewww, 2023-10-05:
Heltec official library dose not support arm64. So we need to first install the Espressif official ESP32 library then add the example code and driver manually.
1. Perquisite:
	* Install Arduino IDE
	*  Install pyserial
		```
		pip install pyserial
		```
	* Add current user to  group dialout and  tty, need reboot/re-login after the usermod command.
		```
		sudo usermod -a -G <group> <username>
		```
2. Install the espressif ESP32 core library 
	* Open Arduino IDE, and click `File`->`Peferences`.
	* Copy the following URL to `Additional Boards Manager URLs`
		```
		https://espressif.github.io/arduino-esp32/package_esp32_index.json
		```
	* Click `Tools` -> `Board` -> `Boards Manager...`, and search `ESP32` in the new pop-up dialog, then select the `esp32 by Espressif Systems ` and click `install`
3. Install the Heltec WiFi LoRa 32 (V3) library
	* Open Arduino IDE, then Select `Sketch`->`Include Library`->`Manage Libraries...` Search `Heltec ESP32` and install it.
4. The examples will be located in `File` -> `Examples` -> `Heltec WiFi LoRa 32 (v3) `
5. Enjoy