Heltec-Aaron-Lee/WiFi_Kit_series #226: WiFi LoRa 32 v3 not working with example code

fbenti, 2023-11-22:
I have  a WiFi LoRa 32 v3 and I uploaded one of the default example  [LoRaSender](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/tree/master/esp32/libraries/Heltec-Example/examples/LoRaBasic/LoRaSender). The upload seems fine as no error is thrown, however, seems like the module enters on a boot loop, and prints the following to the serial port: 

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400454d5
SPIWP:0xee
mode:QIO, clock div:2
load:0x3fce3808,len:0x43c
ets_loader.c 78
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:QIO, clock div:2
load:0x3fce3808,len:0x43c
ets_loader.c 78
```
Can anyone help me solving this issue?

PaulSchulz, 2023-12-11:
> I have been trying for three days now, and nothing works, tried lots of
> websites, and nobody else has them working either. Sorry I bought them,
> it's my second bad experience with this supplier, and there won't be a
> third. If you get it going, would you please let me know. Thanks

Hi BlueSkySunset76,

With the ArduinoIDE, this problem has been fixed for Linux/Ubuntu users
(including myself) by upgrading the esptool package to the most recent
available version.

The fix details are in issue# 159

https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/issues/159#issuecomment-1842269504

BlueSkySunset76, 2023-12-11:
Thanks, I returned them, and will order different type from a different supplier.D:

https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/issues/159#issuecomment-1842269504

BlueSkySunset76, 2023-12-18:
I returned them. I think what they did when they upgraded and made all of the available working (Arduino) code obsolete, shows quite a bit of disregard for the customer. Purchased some Lilygo boards, worked within a half hour of opening the boxes. Working very well. Thanks for your reply though, and offer of assistance D:

https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/issues/159#issuecomment-1842269504

vvaidy, 2023-12-29:
For the benefit of folks that aren't exactly sure what "the fix" referred to above is, here is (an updated version of) the fix that worked for me on Ubuntu/PopOS:

```
mv ~/.arduino15/packages/Heltec-esp32/tools/esptool_py/3.3.0 ~/.arduino15/packages/Heltec-esp32/tools/esptool_py/3.3.0.bak
export ESPTOOL_VER=4.7.0 # update this to a newer version if you like, 4.7.0 worked for me
wget https://github.com/espressif/esptool/archive/refs/tags/v${ESPTOOL_VER}.zip
unzip v${ESPTOOL_VER}.zip
rm v${ESPTOOL_VER}.zip
mv esptool-${ESPTOOL_VER} ~/.arduino15/packages/Heltec-esp32/tools/esptool_py/3.3.0
```
Depending on when you are reading this, you might need to change/update the `3.3.0` above, which you can verify/find by looking at the console log when the arduino IDE starts up.

Hope this helps.

joaopedros2, 2024-01-10:
I'm experiencing the same problem on Windows 10 that was working perfectly fine until a few months ago. I initially thought it might be a hardware issue with the boards, but after testing the sender and receiver sketches, I discovered the problem only happens with the sender. It's unfortunate that there isn't a solution available for Windows users. I'll need to find a way to use Ubuntu specifically for this situation.

vvaidy, 2024-01-21:
> I'm experiencing the same problem on Windows 10 that was working perfectly fine until a few months ago. I initially thought it might be a hardware issue with the boards, but after testing the sender [...]

If you look at the steps above, it's pretty straightforward and you should be able to do the same thing on Windows. In short, it's deleting the 3.3.0 version of esptool, downloading a 4.7.0 version from github and then copying that as if it were the 3.3.0 version ... it's a bit of a hack, but I'm guessing that would likely work on Windows as well?

tanguy-e, 2024-02-08:
I tried the fix but i have an error during compilation whereas all was fine before the fix : 

Traceback (most recent call last):
  File "/home/tanguy-e/.arduino15/packages/Heltec-esp32/tools/esptool_py/3.3.0/esptool.py", line 34, in <module>
    import esptool
  File "/home/tanguy-e/.arduino15/packages/Heltec-esp32/tools/esptool_py/3.3.0/esptool/__init__.py", line 41, in <module>
    from esptool.bin_image import intel_hex_to_bin
  File "/home/tanguy-e/.arduino15/packages/Heltec-esp32/tools/esptool_py/3.3.0/esptool/bin_image.py", line 16, in <module>
    from intelhex import IntelHex
ModuleNotFoundError: No module named 'intelhex'

exit status 1

Compilation error: exit status 1

vvaidy, 2024-02-08:
> I tried the fix but i have an error during compilation whereas all was fine before the fix :
> 
> Traceback (most recent call last): File "/home/tanguy-e/.arduino15/packages/Heltec-esp32/tools/esptool_py/3.3.0/esptool.py", line 34, in import esptool File "/home/tanguy-e/.arduino15/packages/Heltec-esp32/tools/esptool_py/3.3.0/esptool/**init**.py", line 41, in from esptool.bin_image import intel_hex_to_bin File "/home/tanguy-e/.arduino15/packages/Heltec-esp32/tools/esptool_py/3.3.0/esptool/bin_image.py", line 16, in from intelhex import IntelHex ModuleNotFoundError: No module named 'intelhex'
> 
> exit status 1
> 
> Compilation error: exit status 1

Hmm. That's interesting ... I wonder why I didn't hot that problem.

Did you try `pip3 install intelhex` or `python3 -m pip install intelhex` ??

tanguy-e, 2024-02-08:
I installed python3-intelhex using apt and it works fine now!
Thanks
Do you know this bug will be corrected in the official version ?