Heltec-Aaron-Lee/WiFi_Kit_series #248: Error with Wifi Kit 32

DireccionCI, 2024-03-25:
I have this error at the moment that I compile my code, and I don´t know why this appear if I don't still use any LoRa project

Library Heltec ESP32 Dev-Boards has been declared precompiled:
Using precompiled library in c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\esp32
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\loramac\region\RegionEU433.c: In function 'RegionEU433TxConfig':
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\loramac\region\RegionEU433.c:636:71: error: 'LORAWAN_PREAMBLE_LENGTH' undeclared (first use in this function)
  636 |         Radio.SetTxConfig( modem, phyTxPower, 0, bandwidth, phyDr, 1, LORAWAN_PREAMBLE_LENGTH, false, true, 0, 0, false, 3e3 );
      |                                                                       ^~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\loramac\region\RegionEU433.c:636:71: note: each undeclared identifier is reported only once for each function it appears in
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\loramac\region\RegionAS923.c: In function 'RegionAS923TxConfig':
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\loramac\region\RegionAS923.c:699:71: error: 'LORAWAN_PREAMBLE_LENGTH' undeclared (first use in this function)
  699 |         Radio.SetTxConfig( modem, phyTxPower, 0, bandwidth, phyDr, 1, LORAWAN_PREAMBLE_LENGTH, false, true, 0, 0, false, 3e3 );
      |                                                                       ^~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\loramac\region\RegionAS923.c:699:71: note: each undeclared identifier is reported only once for each function it appears in
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\loramac\region\RegionCN470.c: In function 'RegionCN470TxConfig':
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\loramac\region\RegionCN470.c:563:64: error: 'LORAWAN_PREAMBLE_LENGTH' undeclared (first use in this function)
  563 |     Radio.SetTxConfig( MODEM_LORA, phyTxPower, 0, 0, phyDr, 1, LORAWAN_PREAMBLE_LENGTH, false, true, 0, 0, false, 4000 );
      |                                                                ^~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\loramac\region\RegionCN470.c:563:64: note: each undeclared identifier is reported only once for each function it appears in
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\loramac\region\RegionCN779.c: In function 'RegionCN779TxConfig':
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\loramac\region\RegionCN779.c:632:71: error: 'LORAWAN_PREAMBLE_LENGTH' undeclared (first use in this function)
  632 |         Radio.SetTxConfig( modem, phyTxPower, 0, bandwidth, phyDr, 1, LORAWAN_PREAMBLE_LENGTH, false, true, 0, 0, false, 3e3 );
      |                                                                       ^~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\loramac\region\RegionCN779.c:632:71: note: each undeclared identifier is reported only once for each function it appears in
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\loramac\region\RegionAU915.c: In function 'RegionAU915TxConfig':
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\loramac\region\RegionAU915.c:597:72: error: 'LORAWAN_PREAMBLE_LENGTH' undeclared (first use in this function)
  597 |     Radio.SetTxConfig( MODEM_LORA, phyTxPower, 0, bandwidth, phyDr, 1, LORAWAN_PREAMBLE_LENGTH, false, true, 0, 0, false, 3e3 );
      |                                                                        ^~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\loramac\region\RegionAU915.c:597:72: note: each undeclared identifier is reported only once for each function it appears in
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\loramac\region\RegionEU868.c: In function 'RegionEU868TxConfig':
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\loramac\region\RegionEU868.c:666:71: error: 'LORAWAN_PREAMBLE_LENGTH' undeclared (first use in this function)
  666 |         Radio.SetTxConfig( modem, phyTxPower, 0, bandwidth, phyDr, 1, LORAWAN_PREAMBLE_LENGTH, false, true, 0, 0, false, 3000 );
      |                                                                       ^~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\loramac\region\RegionEU868.c:666:71: note: each undeclared identifier is reported only once for each function it appears in
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\driver\gpio-board.c: In function 'GpioMcuWrite':
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\driver\gpio-board.c:54:22: error: 'RADIO_NSS' undeclared (first use in this function)
   54 |         if(obj->pin==RADIO_NSS && radio_nss_inited==false)
      |                      ^~~~~~~~~
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\driver\gpio-board.c:54:22: note: each undeclared identifier is reported only once for each function it appears in
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\loramac\region\RegionIN865.c: In function 'RegionIN865TxConfig':
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\loramac\region\RegionIN865.c:640:71: error: 'LORAWAN_PREAMBLE_LENGTH' undeclared (first use in this function)
  640 |         Radio.SetTxConfig( modem, phyTxPower, 0, bandwidth, phyDr, 1, LORAWAN_PREAMBLE_LENGTH, false, true, 0, 0, false, 3e3 );
      |                                                                       ^~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\loramac\region\RegionIN865.c:640:71: note: each undeclared identifier is reported only once for each function it appears in
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\LoRaWan_APP.cpp:40:34: error: 'LoRaWAN_DEBUG_LEVEL' was not declared in this scope; did you mean 'CORE_DEBUG_LEVEL'?
   40 | RTC_DATA_ATTR uint8_t debugLevel=LoRaWAN_DEBUG_LEVEL;
      |                                  ^~~~~~~~~~~~~~~~~~~
      |                                  CORE_DEBUG_LEVEL
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\LoRaWan_APP.cpp: In member function 'void LoRaWanClass::sleep(DeviceClass_t)':
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\LoRaWan_APP.cpp:666:40: error: 'HELTEC_BOARD' was not declared in this scope
  666 |         Mcu.sleep(classMode,debugLevel,HELTEC_BOARD,SLOW_CLK_TPYE);
      |                                        ^~~~~~~~~~~~
c:\Users\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\LoRaWan_APP.cpp:666:53: error: 'SLOW_CLK_TPYE' was not declared in this scope
  666 |         Mcu.sleep(classMode,debugLevel,HELTEC_BOARD,SLOW_CLK_TPYE);
      |                                                     ^~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

marlontejadat, 2024-03-26:
I have the same issue.

Frosty-STI, 2024-03-26:
Me too!! I can't find any documentation for resolving this anywhere.

whitehat101, 2024-03-28:
I saw this error when using the "Heltec_ESP32 Dev-Boards" library version **2.0.1**. I was able to downgrade to version **1.1.5** and my sketch compiled cleanly.

Mr-HaleYa, 2024-03-28:
They had a bug, its been fixed in this PR https://github.com/HelTecAutomation/Heltec_ESP32/pull/133
You will have to download the repo and manually install it as a library to use the latest until they push a new version with the fix.

valdt, 2024-03-28:
Bug remains even with PR  ? Anyone els having problems ?

DireccionCI, 2024-03-28:
Yes I can remain the problem with that and the other comment behind TSM people

algspd, 2024-04-06:
Same here, using the lib from github

algspd, 2024-04-06:
Solved, add to your boards manager URLS:
https://resource.heltec.cn/download/package_heltec_esp32_index.json

And install:
Heltec ESP32 Series Dev-boards

Then select the board, and it works.

https://docs.heltec.org/en/node/esp32/esp32_general_docs/quick_start.html#via-arduino-board-manager

2ZZ, 2024-04-13:
Same issue here even with the above board manager installed.
Downgrading to 1.1.5 works, thanks for the tip @whitehat101