Heltec-Aaron-Lee/WiFi_Kit_series #249: Error with new Framework v3.0.0

tanguy-e, 2024-03-27:
Hello, updating the framework to 3.0.0 and library to 2.0.1 leads to compilation error : 
In file included from c:\users\tanguy-e\appdata\local\arduino15\packages\heltec-esp32\tools\xtensa-esp32s3-elf-gcc\esp-12.2.0_20230208\xtensa-esp32s3-elf\include\c++\12.2.0\bits\ios_base.h:46,
                 from c:\users\tanguy-e\appdata\local\arduino15\packages\heltec-esp32\tools\xtensa-esp32s3-elf-gcc\esp-12.2.0_20230208\xtensa-esp32s3-elf\include\c++\12.2.0\ios:42,
                 from c:\users\tanguy-e\appdata\local\arduino15\packages\heltec-esp32\tools\xtensa-esp32s3-elf-gcc\esp-12.2.0_20230208\xtensa-esp32s3-elf\include\c++\12.2.0\ostream:38,
                 from c:\Users\tanguy-e\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Strings/JsonString.hpp:8,
                 from c:\Users\tanguy-e\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Strings/Adapters/JsonString.hpp:8,
                 from c:\Users\tanguy-e\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Strings/StringAdapters.hpp:8,
                 from c:\Users\tanguy-e\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Memory/StringPool.hpp:11,
                 from c:\Users\tanguy-e\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Memory/ResourceManager.hpp:8,
                 from c:\Users\tanguy-e\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/JsonVariantConst.hpp:10,
                 from c:\Users\tanguy-e\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:9,
                 from c:\Users\tanguy-e\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/ElementProxy.hpp:7,
                 from c:\Users\tanguy-e\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/JsonArray.hpp:7,
                 from c:\Users\tanguy-e\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:29,
                 from c:\Users\tanguy-e\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from c:\Users\tanguy-e\Documents\Arduino\libraries\CayenneLPP\src/CayenneLPP.h:11,
                 from C:\Users\tanguy-e\Documents\Arduino\LoRaWAN_End_Device\LoRaWAN_End_Device.ino:4:
c:\users\tanguy-e\appdata\local\arduino15\packages\heltec-esp32\tools\xtensa-esp32s3-elf-gcc\esp-12.2.0_20230208\xtensa-esp32s3-elf\include\c++\12.2.0\system_error:39:10: fatal error: bits/error_constants.h: No such file or directory
   39 | #include <bits/error_constants.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1

Mr-HaleYa, 2024-03-28:
They had a bug, its been fixed in this PR https://github.com/HelTecAutomation/Heltec_ESP32/pull/133
You will have to download the repo and manually install it as a library to use the latest until they push a new version with the fix.

dirkbeer, 2024-06-22:
I have this same problem.