Heltec-Aaron-Lee/WiFi_Kit_series #255: PlatformIO VSC config Ubuntu

jo-ei, 2024-05-25:
Hi,
i like to use it on my Ubuntu with Visual Studio Code with platform.io.

here is my platformio.ini config 
```
[platformio]
default_envs = my-config
description = my Test Project

[env:my-config]
platform = https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series.git
board = heltec_wifi_lora_32_V3
framework = arduino
platform_packages =
  framework-arduinoespressif32@https://github.com/HelTecAutomation/Heltec_ESP32.git
monitor_speed = 115200

lib_deps =
  heltecautomation/ESP32_LoRaWAN@^2.1.1
  heltecautomation/Heltec ESP32 Dev-Boards@^2.0.2
```

i got the following error:
```
Resolving my-config dependencies...
Platform Manager: Installing git+https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series.git
git version 2.34.1
Klone nach '/home/user/.platformio/.cache/tmp/pkg-installing-uom9pai3' …

MissingPackageManifestError: Could not find one of 'platform.json' manifest files in the package
```

Can somebody help me to get this fixed?

Thanks

SpaceTeddy, 2024-05-25:
+1
I'm also interested to keep the Heltec Lora V3 running under PLatformIO. Arduino IDE is so lame :)

ryan-lang, 2024-07-26:
@jo-ei Were you able to find a working ini config?

jo-ei, 2024-07-26:
yes i could, but with a different lib from [ropg/heltec_esp32_lora_v3
](https://github.com/ropg/heltec_esp32_lora_v3/issues/24#issuecomment-2212470195)

Perhaps there is a solution here too