Heltec-Aaron-Lee/WiFi_Kit_series #268: OLED issues with Heltec Wifi LoRa v3 (version 3.2) vs Heltec Wifi LoRa v3 (version 3.1)

muoto, 2025-04-21:
I have tried to run the same code on both the v3.1 and v3.2, OLED works fine on 3.1 but is the text is barely visible in 3.2.  I know it's not the display because TinyGS works well on both versions.  Have tried all sorts of things to fix it (changing delay in reset OLED, setWire.setClock(100000), setting SSD1306_SETCONTRAST to max), but it has no effect. Do you have any ideas as to what may be going on?

You can find the version number (3.1 or 3.2) under the Antenna section. Curious if anyone else has run into issues.

quallenhusten-ui, 2025-09-28:
Hi, I´ve got the same problem...maybe something wrong in the unofficial_library.h    when using HT-SSD1306Wire.h it works. not with unofficial lib.....

djcasper1975, 2025-10-26:
add this display.setContrast(255);